supermerill/SuperSlicer #2439: Can't open SuperSlicer. Slicer doesn't open after loading screen.

ItzWilliee, 2022-02-22:
### Version
2.3.58.0
### Operating system type + version
Windows 10, 20H2 / 19042.1526

### Behavior
I was able to open SuperSlicer multiple times but after a few times it failed to open past the loading screen

* _Steps needed to reproduce the problem_
Open SuperSlicer?

* _Expected Results_
Open SuperSlicer

* _Actual Results_
Initial loading screen like normal then disappears like it should but the slicer doesn't open

  * _Screenshots if useful_
https://user-images.githubusercontent.com/66348259/155118730-512fe899-afba-4162-bb32-ae9e8cabdd55.mp4

ItzWilliee, 2022-02-22:
I've tried extracting the zip to the same folder where SuperSlicer is to replace any possible missing files. Also tried extracting to a different drive but neither of these worked.

supermerill, 2022-02-22:
And the nightly? https://github.com/supermerill/SuperSlicer/suites/5387050935/artifacts/169573398

intositeme, 2022-02-22:
I downloaded the release [SuperSlicer_2.3.58.0_win64_220214_bis.zip](https://github.com/supermerill/SuperSlicer/releases/download/2.4.58.0/SuperSlicer_2.3.58.0_win64_220214_bis.zip) And can't startup as well. I deleted the settings I had in AppData and I could start. I tried copying my previous settings, and it won't start again. Then I tried copying only Printer settings, it work, filament settings it work, print settings and it won't start. 

So only used printer & filament settings from previous. Opened a 3mf file used in previous SS version, and it worked, then saved the new print settings. Tried to start and it won't start as well. Could there be an issue trying to load print settings?

intositeme, 2022-02-22:
> And the nightly? https://github.com/supermerill/SuperSlicer/suites/5387050935/artifacts/169573398

This version seems to load up fine with print settings in them. Even loads all my settings from 2.3.57

ItzWilliee, 2022-02-23:
> intositeme

same here the nightly worked perfect and kept my settings

celtare21, 2022-02-23:
This one doesn't start for me

closed 2022-07-14 by supermerill