supermerill/SuperSlicer #2440: Multiple open projects / window / instances [possibly macOS only]

alexvonsiebenthal, 2022-02-22:
**Is your feature request related to a problem? Please describe.**
I am often working on multiple projects and printers in parallel. Hence it would be handy if I was able to keep more than one project/window open. I don't know if it's just a limitation of the macOS version of SS, but I cannot keep two windows open at the dame time.

**Describe the solution you'd like**
see above

**Describe how it would work**
Multiple instances/windows just like in any other app, e.g. a browser or open office

**Describe alternatives you've considered**
Saving, closing, opening, using, saving, closing, opening, ...

supermerill, 2022-02-22:
preference -> allow just a single instance -> uncheck ?

alexvonsiebenthal, 2022-02-22:
> preference -> allow just a single instance -> uncheck ?

Did this. However, the previous window just gets replaced (and disappears) with the new one.
macOS does not work like Linux or Windows where you run multiple instances of a GUI app. The apps are supposed to run once and facilitate an array of windows.

alexvonsiebenthal, 2022-02-22:
> mmm. I wonder if there is the same behaviour with prusaslicer or if I introduce a bug somewhere.

PS does have the same behaviour

supermerill, 2022-02-22:
If you use Window -> open new instance. It still doesn't work?

supermerill, 2022-02-22:
I really fear breaking things, as it's hard for me to test on macos.

alexvonsiebenthal, 2022-02-22:
For instance, if you assign .3mf as filetype to be opened with SS, it'll be confused. macOS works this way: it first looks, if there is an instance of the app and then hands it over to that instance. If that's not the case, it will open a new instance. Hence, when opening a file from the file manager it'll overwrite the file open in one of the instances.

> I really fear breaking things, as it's hard for me to test on macos.

I can understand that. If you want help in testing, I'm happy to assist. I just can't help you with the C(++) stuff :-)

alexvonsiebenthal, 2022-02-22:
Btw: congrats!
[image]