supermerill/SuperSlicer #2441: A debugging check in this application as failed

bergie5737, 2022-02-23:
### Version
_Version of SuperSlicer used goes here (help->about)_2.3.57.11

### Operating system type + version
_What OS are you using, and state any version #s_Windows 11

### Behavior
* _Describe the problem_ Getting pop up that says 'A debugging check in this application has failed' 
* _Steps needed to reproduce the problem_ Mostly when I drag a new 3MF or STL unto the build plate
* _Expected Results_ No errors
* _Actual Results_ Pop up warning me. 
  * _Screenshots if useful_ 
[image]

_Is this a new feature request?_ No
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
[Hard holder.zip](https://github.com/supermerill/SuperSlicer/files/8121001/Hard.holder.zip)

amsmith525, 2023-11-04:
I have a similar error -- I notice in the first post as well as mine the path appears to be incorrect -- "\a\SuperSlicer\SuperSlicer\"

[image]

amsmith525, 2023-11-04:
This occurred while dragging a .gcode file into the program window

supermerill, 2023-11-22:
For the first one, I don't have enough detials on how to replicate (also, the version is a bit old, maybe it's already fixed).
For the drop, fixed in next release.