supermerill/SuperSlicer #2444: A question about seams

poloniumn, 2022-02-23:
Using seam painting and selecting aligned does not align.
[image]
[image]

I tried seam spheres, but it can't accurately place seams at multiple specified locations, the spheres will attract seams that should be the responsibility of other spheres, I tried for a long time and didn't find a suitable attraction distance, is the software unable to implement the function of placing aligned seams at specified locations?

If it's not too difficult to implement, having both alignment and  seam painting have an effect will make seam control a breeze.

supermerill, 2022-02-23:
You're on 2.3.57.11, right?

The aligned seems not aligned enough...

supermerill, 2022-02-23:
Oh, I understand.
If you paint, that takes precedence over the "aligned". Try to paint only at the first layer(s), the one over them should use the aligned method to stay on top.

poloniumn, 2022-02-23:
Yes, I am using 2.3.57.11.

I selected aligned in seam position and drew a line using seam painting, the seams would be messily distributed on the inside of the line, not even in the specified area, I needed this to limit seams running amuck and even then it wouldn't align.
[image]

poloniumn, 2022-02-23:
Oh, and one more question, in the Gcode preview the software doesn't seem to show the seams?

I remember that in Prusa seams are clearly marked in white, but here I can only use Retracrions to see where the seams are, and it doesn't work well, is there some configuration I haven't turned on?

Fredy12335, 2022-02-23:
Hi!
While we are on the subject, allow me to comment. It's a great idea to have a hand-drawn seam, but a useful feature would be a line drawing tool in addition to the brushes. Like in paint. Because I can't draw a straight line by hand. 
In some cases I solve this by turning on move the mouse cursor with the keyboard in windows but this is not a perfect solution.
[image]

supermerill, 2022-02-23:
>  the software doesn't seem to show the seams?

In the 2.4, it does.

> I can't draw a straight line by hand.

use a seam modifier shpere. When there are at least one, it will set the seams on the nearest one. (nearest in Z, then nearest in XY).It's easier to get strait lines.

poloniumn, 2022-02-23:
> Hi!
> While we are on the subject, allow me to comment. It's a great idea to have a hand-drawn seam, but a useful feature would be a line drawing tool in addition to the brushes. Like in paint. [...]

The seam spheres are great when you just need a specified seam position. 

My problem is that I have to hide my seams in many places on the model and seam spheres are just too difficult to do.

Jamin82, 2022-03-29:
> Use a seam modifier sphere. When there are at least one, it will set the seams near the nearest one. (nearest in Z, then nearest in XY). It's easier to get strait lines.

I am unable to get seams to align with the sphere. it worked fine in 2.3 but not in any of the 2.4 nightly builds