supermerill/SuperSlicer #2445: Filament -> Cooling missing

intositeme, 2022-02-23:
### Version
Version: 2.4.58.0
Build: SuperSlicer_2.4.58.0_2022-02-21

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 22000), 64-bit edition

### Behavior
Filament Cooling section is missing. 
*  Load any 3mf file
*  Cooling section to be displayed in expert mode under Filament settings.
*  Cooling section is not being displayed
[image]

supermerill, 2022-02-23:
fixed in Build: SuperSlicer_2.4.58.0_2022-02-23

supermerill, 2022-02-23:
you can also delete the line `group:Behavior`  in the filament.ui file.

intositeme, 2022-02-24:
> you can also delete the line `group:Behavior` in the filament.ui file.

Thanks, but where do I find it?

supermerill, 2022-02-27:
resources/ui_layout

closed 2022-04-01 by supermerill