supermerill/SuperSlicer #2446: Scaling factor for flow with line width

blex-max, 2022-02-23:
**Is your feature request related to a problem? Please describe.**
A follow up from #2028: 

Me:
> What I'm finding as I experiment with this [solid infill line overlap] is that flow, unsurprisingly I think, doesn't vary linearly with line width. I think the reason I needed this feature is that I have large line widths for my perimeters (c.a. 150% of nozzle 0.6mm diameter) compared to my solid infill (c.a. 105%) (I'm using beefy nozzles/heaters/motors of course). In other words, while this new feature for solid overlap solves my problem, it's sort of a bandage on the underlying issue - **I think the 'real' solution would be to introduce some kind of scaling factor for flow as a function of line width.** There would even a way to calibrate it per this CNC Kitchen [video](https://www.youtube.com/watch?v=UNJdv5bFGOg&t=499s) This has been sort of attempted in Marlin itself with the scaling factor for PID temp as a function of extrusion volume. What do you think? Perhaps I could fork and work on it?

@supermerill:
> For the mm3/sec -> mflow mult relation, the problem when working on it on the slicer side is that we don't know the real speed (so same for the flow-speed).
> Ideally, you have to implement it in the firmware side.
> The slicer can help you with calibration prints.

I've been thinking this over and poking at Marlin, and I actually don't think this should be implemented firmware side - for one thing it changes on a per filament basis. I agree that it would be very difficult to implement _in relation to mm3/sec_, but actually I don't think we need to know anything about the mm3/sec. I think the implementation can just be a linear scaling factor that changes flow with line width. Then it can be naively calibrated either by result/eye on a series of test prints, or by extruded weight per that CNC kitchen video.

**Describe the solution you'd like**
A naive scaling factor for flow with line width that is filament specific.

**Additional context**
I'm willing to work on this myself if you're unconvinced (and I'm kind of keen to), it might take me a while to understand where to start though!

FDResearch, 2022-02-25:
I was just trying to mess with settings to fix this exact problem so I would like to second this or some similar solution, at the very least giving better options to increase or decrease flow based on feature type would be great, then at least I can change them in the print profile without having to input incorrect extrusion calibrations. 

Also either option (fixed flow factor change based on feature type or width based scaling factor) should probably have settings for infill and perimeters as it seems the perimeters are affected worse. Once I crank the extrusion multiplier high enough to get solid perimeters with 1.2mm x 0.16 on a 0.6mm nozzle the solid infill is over extruding (even though it was under extruding with the "correct" multiplier that gives perfect results on reasonable widths).

This issue of under extruding with wide extrusion widths also seems to scale with layer height to some degree.

And to the answer of why would you need 1.2mm+ extrusion widths of course. Both mine and numerous other testing has shown significant improvements in layer bonding with such high aspect ratios, I'm seeing improvements in my IZOD impact testing of layer bonding even with gaps between perimeters due to under extruding.

blex-max, 2022-02-25:
@FDResearch you're in exactly the same boat as I was! If you go thru the issue I reference, Merrill was kind enough to implement 'solid infill line overlap' in the 2.4 beta - check that out as it may alleviate some of your issues

FDResearch, 2022-02-25:
I just installed it after posting that actually and that will help, but having more fine tuned control would be great, it may be possible to do a hacky work around using the gcode between extrusion role change and the flow modifer gcode but I'm not exactly sure how to code that. I'm also trying to fix gap fills that are not fully filled which having a control on flow based on features would also solve. But this is on parts with reasonable extrusion widths so think I can just use the solid fill overlap to tone solid fill down and up the extrusion multiplier.

supermerill, 2022-03-07:
Input if you want to work on it:
 * GUI: try to replicate the window for ramming settings. That should create a string with all the value needed inside.
 * gcode: in the gcode, you have to un-serialize the string to get useful numbers (create a small class for that, that init itself with the string and can "modify" a flow given parameters. Save it inside the gcode object
  the best part to init it is when the extruder change. (so in gcde::set_extruder)
  then call the method for flow change at the start of Gcode::_extrude (when you set e_per_mm)