supermerill/SuperSlicer #2449: Klipper Physical Printer

eknowlton, 2022-02-24:
### What happened?

I went to add a physical printer and when I selected my MK3S+, and then I selected Klipper, it gave me the correct fields and I added my printer and that was a success.

Then I went to add my Voron 2.4 physical printer with the same process, although when the dialog to add a physical printer loaded, it had Duet pre-selected, and when I switched it to Klipper, I was presented with the options for Repetier server ( I believe, just not Klipper options ).

I expected it to present options for connecting to a Klipper instance. 

### Project file & How to reproduce

1. Create Voron 2.4 Printer
2. Select Voron 2.4 Printer
3. Go to Printer Settings
4. Select "Add Physical Printer" COG Icon Button
5. Should be presented with "Duet" selected.
6. Switch to "Klipper"
7. Encounter bug

### Version

2.4.58.0

### Operating system

Ubuntu 21

### Printer model

Voron 2.4 with Afterburner

eknowlton, 2022-02-24:
Also want to say I love this slicer, and really appreciate the work. Maybe I can help one day.

supermerill, 2022-03-07:
Can you try with the new beta?