supermerill/SuperSlicer #2450: First layer -> travel speed

StefanMeGit, 2022-02-24:
**Is your feature request related to a problem? Please describe.**
Layer adheason by travel speeds while hitting the material with the nozzle on high speed printers with travel moves with 500mm/s 

**Describe the solution you'd like**
Just an easy field to input a travel speed for the first layer (mm and % in respect to the default travel speed)

**Describe how it would work**
---
**Describe alternatives you've considered**
---
**Additional context**
---

Just miss this feature really hard. I think its an easy feature which helps a lot for fast printers. 

Thank you

supermerill, 2022-03-07:
duplicate of #2204

SamSpiri, 2022-03-14:
Duplicate of #1339