supermerill/SuperSlicer #2452: Add Volumetric speed limit to Extruder

fred099, 2022-02-24:
**Is your feature request related to a problem? Please describe.**

I really like the Autospeed feature and have adopted it more or less completely. Using relative acceleration limits and volumetric speed is one of the best new features in my opinion.

There is one small detail that stops me from being able to use the same print profile between my printers, and that is where you set the Volumetric speed limit. Currently it sits within the Print Settings/Speed.

**Describe the solution you'd like**

I'd like to move the Volumetric speed limit from the Print profile and make it a limit related to the Extruder in the Printer setting. 

**Describe alternatives you've considered**

An additional Volumetric speed limit in the Extruder section that would supersede both the limit in Printer settings and Filament settings.

**Additional context**

This change would make it possible (however perhaps not optimal) to use the same Print profile between very different printers.

kursatu, 2022-02-28:
This was brought up before and agreed by all. I can certainly use it.
Otherwise, I need to change it when I change my dragon high-low flow nozzles with combinations 
of bondtech CHT nozzles and regular nozzles. 
https://github.com/supermerill/SuperSlicer/issues/491

strayr, 2022-02-28:
This is where it gets conceptually difficult. You'll still have different max vol flows wit each filamant with a high flow hotend and s the slow rate is a combination of the filament and hotend/nozzle.

What works for me is having a separate filment profile for my highflow setup, as I'm likely cooling it more agressivley as well when spewing filament at 35mm3/s instead of 11mm3/s and my pressure advance settings have changed, althogh i'm handling pressure advance settings across a few nozzle sizes and between a bowden setup and several direct drives by using custom gcode and looking for keywords.

What's missing and may be unsolvble is a a way of tracking the union of filamant and hotend configurations other than duplicating filament settings.

bartlammers, 2022-05-20:
I would also really like this feature. Even if it just an upper limit, and not taken into account when calculating auto-speeds of any kind.
For multiple printers I like to use the same profiles, both for filament and print settings. One can do 20+mm³/s though and the other one tops out around 12

EvilGremlin, 2022-07-24:
Volumetric limit depends on temperature so it's correct to place it in filament profile. Extruder setting should exist too as a multiplier of filament value, because this is how it works. Also there should be two volumetric values - one for shell and other for everything else.

supermerill, 2022-07-24:
kinda related to #491