supermerill/SuperSlicer #2455: Extrusion Multiplier only works for 1 and .95 values, ignores other

ModMike, 2022-02-25:
### What happened?

When printing an Ellis flow tile test with values from .90 to 1.0, only the .95 an 1.0 tiles are different. I redid the test with the wall thickness technique, and same result.

### Project file & How to reproduce

[Extrusion Multiplier.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8137911/Extrusion.Multiplier.3mf.zip)

### Version

2.3.57.11

### Printer model

Ender 3 but Voron users also reporting issue in discord.

supermerill, 2022-03-07:
I don't understand.

ModMike, 2022-03-07:
When setting the extrusion multiplier to anything other than 1 or .95, there is no effect because one of the libraries from Slic3R is not working properly. I forgot the name of the library but will figure it out and update this post.

In order to get the extrusion multiplier to somewhat work, I needed to set it to .95 and then set default width to 111%. This is a hack that is going around in the Voron community to get it to work.

This has been verified by a few people, notably Martin whom I believe contacted you about it.

supermerill, 2022-03-07:
If I set the extrusion_multiplier to 2, I have 2 times the length of filament used in the gcode file.
So by reading "Extrusion Multiplier only works for 1 and .95 values, ignores other", I  clearly have a problem understanding what you really means by that.

ModMike, 2022-03-07:
I updated the the first to make it easier to understand and reproduce.

legend069, 2022-03-08:
> You have to print to replicate. I will upload pics later today.
> 
> Thank you for taking the time to look at this!

providing the sliced g code will be fine, as we can see the output values for each extrusion,
if you have problems with it printing it could be several other things.

please provide the sliced g code file for extrusion multiplier set to 1.0, 1.05, 0.95
or the specific values you think are causing problems.

supermerill, 2022-03-08:
Also "Measure wall thickness" is an unreliable metric.

ModMike, 2022-03-14:
I gave up on this seeing as Sli3r and Prusa said they would look at it. By the looks of it there was a new build that specifically addresses this, not sure if it's the issue I was originally referring to. By the way, the latest Mac nightly build is corrupt for some reason.