supermerill/SuperSlicer #2458: External Perimeter 1st: Re-order Walls for Overhangs

CCS86, 2022-02-25:
The option to print external perimeters is great in most situations. You get a much cleaner placement of that wall and external finish, since there is no accumulation of wall defects (over-extrusion, ripples, etc) to print through by pushing on that external wall.

The main exception to this seems to be when there is a significant overhang. Without another wall to stick to, that external wall can print in mid-air. 

My feature idea would be a threshold that detects whether a given layer contains overhanging regions. If the amount of overhang crossed this threshold, it would change the wall printing order. If the external wall was #1, then #2, #3, #4, as we go inwards on a normal layer; it would print #2, #1, #3, #4, #5 on layers with enough overhang,  

You could potentially take it a step further if there was extreme overhang and print #3, #2, #1, #4, #5, #6.

liftbag, 2022-02-25:
I am afraid that the change of order can create artifacts, for example a visible line at the layer where the change occurs. Something similar to what happens with the ventilation automatisms activated on the 3Dbenchy print.
It would be interesting to try it though, because the problem you exposed is real, and the only way currently to try to solve it is to manually change the order of the outer perimeters using the layer range madifier, but doing so only changes the order of the outer layer.

CCS86, 2022-02-25:
I don't doubt that it could create artifacts. But, the reality is that the current default behavior (outer wall last) creates tons of artifacts too. Outer wall first reduces this quite a bit, but can ruin overhangs.