supermerill/SuperSlicer #2459: 100% first layer height not allowed?

remizik, 2022-02-25:
### What happened?

Hi,
I just updated to version 2.3.57
My profiles remained unchanged, great.
But I can't slice anymore with the error message : "first layer height can't be greater than max layer height"
These are the settings I think are relevant ?

[image]

[image]

I'm getting these messages : 

[image]

### Project file & How to reproduce

Set the settings as described above

### Version

2.3.57

### Operating system

windows 10

### Printer model

ender 3

liftbag, 2022-02-25:
Better to specify the measurement in mm rather than the percentage. The percentage applies to the nozzle diameter. If you have indicated a 0.4 nozzle in the printer tab, that 100% means 0.4mm.

remizik, 2022-02-25:
> Better to specify the measurement in mm rather than the percentage. The percentage applies to the nozzle diameter. If you have indicated a 0.4 nozzle in the printer tab, that 100% means 0.4mm.

ohhh my bad I didn't read the description popup enough.. I thought it was a %age of the "base layer height" value right above..
oops!

Thanks a lot for your answer

remizik, 2022-02-25:
I can't read.

closed 2022-02-25 by remizik

supermerill, 2022-03-07:
It's the % of base layer height in old PrusaSlicer (and old slic3r I think).
But that means you can't change the layer height. It's bad. That maybe why prusa just remove the % option for this field.