supermerill/SuperSlicer #2462: Feature Request: Monotonic perimeter

mirasderbissalin, 2022-02-26:
Seams are inevitable. You can hide them, sure,  but that's not always possible, especially for cylindrical objects/features.

In the picture below on the left is a conventional way of putting down a layer of two perimeters. Put inner perimeter first (usually), stops extrusion then move 0.4mm (center-to-center distance for 0.4mm extrusion) to the right and put down the outer perimeter. 

On the right (and what I purpose to implement) printhead puts down inner perimeter first  (moving CCW) stops ~2mm before finishing that perimeter, moves 0.4mm (center-to-center distance for 0.4mm extrusion) to the right WITHOUT stopping extrusion, lays down the outer perimeter (CW), finishes it to the steps, moves 0.4mm (center-to-center distance for 0.4mm extrusion) to the left WITHOUT stopping extrusion and finishes that inner perimeter.
[image]

Aligning seams is a little tricky now in SS, and filing the seam is not always possible (with nylon, FE) and is never fun.

With greatly tuned kinematics seam is about the only thing slicing-wise, that throws the dimensions away.

Have a great day!

mirasderbissalin, 2022-02-26:
Also, something like that can probably reduce cornering on the sharp turns
[image]

neophyl, 2022-02-26:
Similar to the function/option that is already there called one loop perimeters ?  Its under Print Settings>Perimeters & Shell>Advanced.  Must be in advanced or expert mode to see the setting.

mirasderbissalin, 2022-02-26:
> Similar to the function/option that is already there called one loop perimeters ? Its under Print Settings>Perimeters & Shell>Advanced. Must be in advanced or expert mode to see the setting.

Oh, didn't know that. I remember seeing that feature, but never investigated it.
Tried it just now, it forces external perimeter to be printed first for some reason. Seem is still there, but now rounded  to inside, that's much nicer, thanks

[image]

mirasderbissalin, 2022-02-28:
I researched a bit more. Turns out that extrusion move is patented by... Stratasys. SOB

CCS86, 2022-03-01:
> Edit: visually seem is more visible. I attribute that to external perimeters printing first

I don't think printing external perimeters first inherently makes them more visible... if anything, it's the opposite. With seam position aligned between inner and outer walls, any seam bulge will stack and become progressively worse, for each wall.

mirasderbissalin, 2022-03-01:
> I don't think printing external perimeters first inherently makes them more visible

oh, yeah, oc. It's just my setup is particularly leaky (0.6, volcano, 255PETG), so the first extrusion move has a gap in the beginning, extra unretract doesn't help, bc it is supposed to be a function of a time a nozzle has been suspended in mid-air. Sorry, forgot to mention that

supermerill, 2022-03-01:
So everything is good?