supermerill/SuperSlicer #2464: FR: Adding brim ears manually

alexvonsiebenthal, 2022-02-26:
**Is your feature request related to a problem? Please describe.**
I haven't found a github issue re manual brim ears, hence this feature request.

**Describe the solution you'd like**
Ideally, I'd be able to add/remove brim ears by hand. Some models just have some funky geometry and SS struggles to find the right brim ears. 

**Describe how it would work**
Ideally, klick on a corner to add/remove brim ears

**Describe alternatives you've considered**
I usually place "fake brim ears" using a 1 layer cylinder. However, this comes at the disadvantage of losing the 'brim offset', hence fusing the model together with the fake ears.

supermerill, 2022-03-01:
A kind of "brim gizmo", you paint where the brim should go?

alexvonsiebenthal, 2022-03-02:
Absolutely. That would be really neat. As long as it also affects the brim ear mode and not just standard brims, that would be excellent.

edddeduck, 2022-08-13:
I'd also like to add my voice to the usefulness of this command. I've recently had to print a few items that could really help with this, some parts really need a brim, but others are areas that removing the brim afterwards is hard so you definitely don't want a brim added. If you could paint the areas that brims should be allowed or added that would be super useful.