supermerill/SuperSlicer #2468: Unwanted supports on top of prints

M4rkusHartmair, 2022-02-27:
### Version
2.3.57.11

### Operating system type + version
Arch Linux 5.16.11-arch1-1 x86_64

### Behavior
Superslicer generates supports on top of printed structures even though "Support on build plate only" is selected.

* _Steps needed to reproduce the problem:_
slicing with option "Support on build plate only" enabled.
occasionally it works as expected, most times there are at least some lines generated on top of printed structures.

* _Expected Result:_ 
No supports on top of printed structures. 
example: Circle of 2cm diameter on build plate 
=> cylindrical area with 2cm diameter over the circle free of any support structures.

* _Actual Results:_
Support lines are generated on top of printed structures, similar but NOT the same as when "Everywhere" is selected for supports.

[image]
[image]

#### Project File (.3MF) where problem occurs
[support_problem.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8149009/support_problem.3mf.zip)
(just renamed!)

supermerill, 2022-03-07:
fixed in 2.4

M4rkusHartmair, 2022-03-12:
Fix confirmed with Version 2.4.58.0 (AUR package aur/superslicer-prerelease-bin 1:2.3.58.0-1)
Supports on both examples now work as expected.

closed 2022-03-12 by M4rkusHartmair