supermerill/SuperSlicer #2473: Crashing at startup. Have to delete config files to open.

leetwanker, 2022-02-28:
### What happened?

Slicer doesn't open. When starting from CLI I get the following information.

```
$ ./SuperSlicer-ubuntu_18.04-gtk3-2.3.58.0.AppImage 
[2022-02-28 17:52:55.310634] [0x00007f8611de85c0] [trace]   Initializing StaticPrintConfigs
[2022-02-28 17:52:55.380566] [0x00007f8611de85c0] [error]   /tmp/.mount_SuperSpsTj1w/bin/../resources/ui_layout/colors.ini
[2022-02-28 17:52:55.380954] [0x00007f8611de85c0] [error]   /tmp/.mount_SuperSpsTj1w/bin/../resources/ui_layout/colors.ini
[2022-02-28 17:52:55.381353] [0x00007f8611de85c0] [error]   /tmp/.mount_SuperSpsTj1w/bin/../resources/ui_layout/colors.ini

[2022-02-28 17:52:55.893796] [0x00007f8611de85c0] [error]   Unable to load bed texture: C:\Manual_Install\Prusa-Slicer Custom Beds\V2_bed_smooth_pei.png
[2022-02-28 17:52:55.893809] [0x00007f8611de85c0] [error]   Unable to load bed model: C:\Manual_Install\Prusa-Slicer Custom Beds\voron2_350bed.stl
Segmentation fault (core dumped)
```
The first time I downloaded 2.3.58.0 and ran it I got this. I suspected it may be an incompatibility with my previous version config files being that it was complaining about not being able to load bed models. So I deleted ~/.config/SuperSlicer and started fresh. That's been a few days ago now, and I'm again crashing at startup and again, deleting ~/.config/SuperSlicer allows SS to start. 

I'm not sure what change I made to cause the new config files to start crashing SS, but I wasn't doing anything weird. All I've done is import Ellis's PIF profiles. 

### Project file & How to reproduce

Unknown.

### Version

2.3.58.0

### Operating system

Linux Mint 19.3

### Printer model

Voron 2.4 with StealthBurner

woschtkopp, 2022-02-28:
Happens to me, too. On Windows platform

Kaden116484, 2022-03-01:
I'm having the same issue, No printing for the night i guess.

tome9111991, 2022-03-01:
Please try the nightly
https://github.com/supermerill/SuperSlicer/actions/runs/1884964972

supermerill, 2022-03-01:
If the problem persist with the nightly, can you zip & post your configuration directory here?

Kaden116484, 2022-03-01:
it's still happening, we are talking over in discord i believe.

woschtkopp, 2022-03-02:
For me the nightlY is working

Katzekonig, 2022-03-03:
Nightly solved it for me. Thanks!

closed 2022-03-03 by Katzekonig