supermerill/SuperSlicer #2474: The lines of the bed can be seen on the object

mojamuto, 2022-03-02:
### What happened?

[image]

The lines of the bed can be seen on the object
this problem was not in version 2.4.58.0, it is new in version 2.4.58.1

### Project file & How to reproduce

[image]

### Version

2.4.58.1 (last nightly) C/C++ Nigthly ubuntu-GTK3 & GTK2

### Operating system

Void Linux

### Printer model

Prusa Bear MK3

liftbag, 2022-03-03:
It's an update from the latest nightly that shows the grid under the bed, but there seems to be a problem. Also present in macOS.

tome9111991, 2022-03-07:
same on win10 58.1 RC
[image]

supermerill, 2022-03-07:
weird, it should only do that for the voron0 (a new feature that had a buggy opengl side-effect).
I tried multiple printers and I don't have that.

supermerill, 2022-03-07:
It's only on linux/ macos? or also on windows?

liftbag, 2022-03-07:
> Oh, understood. It's also triggered for custom printer with a texture.

Can you guide me where to set to bed_with_grid = 0? I never set a Voron printer. I only have set a custom printer with Lerdge firmware.
The problem persists even with the latest nightly RC 2.4.1.

supermerill, 2022-03-07:
> Can you guide me where to set to bed_with_grid = 0? I never set a Voron printer. I only have set a custom printer with Lerdge firmware.

for custom printers, it's a typo in 3dBed.cpp line 298, true instead of false.

supermerill, 2022-03-07:
will be in nightly in some hours

supermerill, 2022-03-07:
Did the nightly resolve the issue?

liftbag, 2022-03-07:
> Did the nightly resolve the issue?

Yes, solved. Thanks!

mojamuto, 2022-03-08:
solved thank you very much

closed 2022-03-08 by mojamuto