supermerill/SuperSlicer #2475: Brim "Ears" not placed in wrigh place.

gin8330, 2022-03-03:
### Version
Version 2.3.57

### Operating system type + version
Windows 11

### Behavior
Brim "Ears" not placed in wright if you copy the same part or "Add" another part. 

[Screenshot .zip](https://github.com/supermerill/SuperSlicer/files/8177137/Screenshot.zip)

supermerill, 2022-03-07:
first layer:
[image]
your part isn't flush with the bed:
[image]

gin8330, 2022-03-07:
Does this part for print not flatten the printing plate horizontally ? Do
you mean that was an issue ?

supermerill, 2022-03-08:
your object is not flat.
It doesn't touch the bed
It's the issue

mrx23dot, 2026-10-28:
please close ticket