supermerill/SuperSlicer #2476: part scale issues

gin8330, 2022-03-03:
### Version
Version 2.3.57

### Operating system type + version
win 11-64 bit
[Read me.zip](https://github.com/supermerill/SuperSlicer/files/8179558/Read.me.zip)

gin8330, 2022-03-07:
> Original size "Roof support" stl is not ment to fil printing plate.
> Srink size to 10% and print. You will se, size just not maching and only
> part
> of the part will be printed...tips just gone.
> Strangly, this time ends of the part was identified correctly and "ears"
> was generated on the pard ends.
>
> Do the same in "Cura" and everything is as should be.
>
> I don't understand
> [image: image]
> <https://user-images.githubusercontent.com/6536403/157070806-4314c5fa-717c-479a-b248-7df9c59f699f.png>

Hello,
would you be kind and explain what exactly confuses you.
Sincerely
Gintas

supermerill, 2022-03-07:
From what you say there is the tip missing, but I don't see any tip.
For the ears, it's because you have selected brim ears.
what is the problem?

gin8330, 2022-03-07:
" From what you say there is the tip missing" -when printer printed part
tip was missing.
Like this:
[image: 1 .png]
And yes, brim it was selected as a brim ears. This time ,same how "brim
ears " war pleased right.
[image: 2 .png]
But it was like that:
[image: Screenshot .png]
But I suppose that happened ,because the part was not flat with the
printing plate.

supermerill, 2022-03-08:
What I see from your last post:
[image]