supermerill/SuperSlicer #2480: Change in "empty layers" warnings from 2.3.57.10 to 2.3.57.11

ornamentist, 2022-03-06:
### What happened?

System: macOS 12.x on an M1 mac.

Since installing 2.3.57.11 most, but not all, STL's I download from Thingiverse, Prusa Printers etc report a warning of empty layers in the model. The same STL's seem to slice fine in Prusa Slicer and Cura.

Looking at some of the models, they appear to be floating above the bed, even after dropping them to the bed. Some that appear on the bed report negative z position values. Repairing the meshes doesn't seem to help.

Falling back to 2.3.57.10 is a good work-around as the problems go away. I haven't tried the 2.4 betas yet as there are no nightly macOS M1 builds and my builds from source are failing (but getting closer).

### Project file & How to reproduce

[gland.zip](https://github.com/supermerill/SuperSlicer/files/8191949/gland.zip)

### Version

2.3.57.11

### Operating system

macOS 12.x

### Printer model

Vivedino Troodon 400 with DDE

supermerill, 2022-03-06:
I fixed a shrinking bug in the .11
You set your vertical hole shrinking compensation to enlarge the holes by 200mm, so when it detect a round hole, it enlorge it so much that all the part is erased.

ornamentist, 2022-03-07:
Good catch!

closed 2022-03-07 by ornamentist