supermerill/SuperSlicer #2481: Help!!! I can't open the project!

CuredPrusa, 2022-03-06:
### Version
_Version of SuperSlicer used goes here (help->about)_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

mojamuto, 2022-03-06:
I think the bug is clear, you are using a 3mf that uses a fill type that does not exist in superslicer/prusaslicer.
That 3mf is created with another program, try to open it with cura for example.
without the file little else can help you

CuredPrusa, 2022-03-06:
nope.
created by and saved in SS.
if you can somehow search the bug inside this 3mf I can send it to you via email. 
I can't even open it in Windows 3D software. There's error in file.

supermerill, 2022-03-06:
If it's not protected, you can post a zip of it here. If not, you can send it to me via discord.

CuredPrusa, 2022-03-06:
Thanks Meril! How to use discord?

CuredPrusa, 2022-03-06:
Ok, i'm just opening the account right now....

CuredPrusa, 2022-03-06:
Ok Meril, but now I'm really stuck. 
I have created my server on discord and I don't know what to do next to send you a file.

CuredPrusa, 2022-03-06:
https://discord.gg/RF6eBRWC

the invitation to my server

CuredPrusa, 2022-03-06:
[image]

CuredPrusa, 2022-03-06:
All parameters were without noticeable error, all settings were appropriately saved, then the project was saved.

mojamuto, 2022-03-06:
I don't know if you have been able to enter discord yet but the superslicer server is located here: https://github.com/supermerill/SuperSlicer/issues/611#issuecomment-907833287

CuredPrusa, 2022-03-06:
Thanks a lot mojamuto. 
I've just entered. 
Let's see what will happen....

CuredPrusa, 2022-03-06:
Thank you Merill for soving this case for me!

closed 2022-03-06 by CuredPrusa