supermerill/SuperSlicer #2482: Failure to load file

SimonSolar2C, 2022-03-06:
### Version
_Version of SuperSlicer used goes here (help->about)_
Version: 2.3.56.8
Build: SuperSlicer_2.3.56.8
### Operating system type + version
_What OS are you using, and state any version #s_
Windows10 64bit

### Behavior
* _Describe the problem_ConfigBase::set_deserialize() failed for parameter "fill_pattern", value ""
* _Steps needed to reproduce the problem_Open the 3mf project file results in error message. I dont know what version saved this file
* _Expected Results_Project opens and loads
* _Actual Results_Error message
  * _Screenshots if useful_
[image]

_Is this a new feature request?_No
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here
[V5e2Finalx2BW.zip](https://github.com/supermerill/SuperSlicer/files/8190631/V5e2Finalx2BW.zip)
_

mojamuto, 2022-03-06:
It seems to be the same problem: #2481

SimonSolar2C, 2022-03-06:
Yes probably the same problem (slightly different message) although it seems it was closed without a solution being posted.

supermerill, 2022-03-06:
probably duplicate of #1983