supermerill/SuperSlicer #2483: Stealth mode generates "M73 Q" that is not supported by Marlin

pedrolamas, 2022-03-07:
### What happened?

Marlin/Klipper G-code output shows `M73 Q` commands that are not supported by Marlin.

This is causing issues in Klipper, as it forces the reported process to always be 0!

### Project file & How to reproduce

- Open "Printer Settings" in SuperSlicer
- Set "Firmware: G-code flavor" to "Marlin" or "Klipper"
- Enable "Firmware: Support stealth mode"
- Enable "Firmware: Print remaining times"

[image]

- Slice any file and save the output G-code
- Open the file in a text editor and search for the M73 entries

[image]

### Version

2.3.57

### Operating system

Windows 11 Pro (21H2) 64bit

### Printer model

Creality Ender-3 V2 with Klipper firmware

neophyl, 2022-03-07:
If you turn stealth mode off does it remove the problematic M73 ?  I'm pretty sure Prusa added that tickbox specifically for their printers that do support it.

pedrolamas, 2022-03-07:
> If you turn stealth mode off does it remove the problematic M73 ?

Yes, disabling the "stealth mode" will only get the first of the `M73` command, and that one is fine as it is Marlin compliant.

> I'm pretty sure Prusa added that tickbox specifically for their printers that do support it.

I agree, as I noticed that it is enabled by default on the MK3 printer profile!

For the record, I did open this same issue in PrusaSlicer repo (https://github.com/prusa3d/PrusaSlicer/issues/8036), but as SuperSlicer has a specific "Klipper" g-code flavor, I do believe that it should not output the problematic `M73 Q` command for it!

neophyl, 2022-03-07:
I may be being a bit dense here but if you can turn the option off then why is it a bug ?  
By enabling the option you are saying your setup supports that functionality.  It would seem more like a misconfiguration option than anything else.

Also I thought one of the strengths of something like klipper was its extensible ?  So if in future klipper expands to add support then it could make use of it.  In the meantime though just turn it off if your printer doesn't like it.

pedrolamas, 2022-03-07:
> I may be being a bit dense here but if you can turn the option off then why is it a bug ?

Granted, the classification as "bug" might depend on the perspective. My personal perspective is that if one sets the G-code flavor to "Marlin", it should only output Marlin compliant g-code (and an `M73 Q` is not part of Marlin g-code).

> Also I thought one of the strengths of something like klipper was its extensible ?

Klipper has support for `M73` commands and uses those to set the print progress, by reading the `P` parameter; however, when it can't find the `P` parameter (such as is when reading the 2nd line), it sets progress to 0.

Sure, one could consider this behavior to be Klipper bug, but then again Klipper is using Marlin compliant code only...

supermerill, 2022-03-07:
Doesn't klipper user can create a macro to intercept a M73?
Maybe someone may want to replicate the noisy/silent of the mk3 in klipper ?

neophyl, 2022-03-07:
The problem with that perspective is that the Prusa Printers use Marlin as type (called Marlin (Legacy) now to differentiate between more recent versions)  so it is a valid gcode for marlin.

In fact I am running a marlin firmware on my Cr10 and sending a M73 with the stealth option on DOES NOT cause any problems on that printer.  Firmware should be programmed to ignore anything it doesn't recognise,  as the stock marlin seems to.

If anything this should be raised as a Klipper firmware issue.

pedrolamas, 2022-03-07:
> Doesn't klipper user can create a macro to intercept a M73?
> Maybe someone may want to replicate the noisy/silent of the mk3 in klipper ?

That sounds reasonable to me.

However, I'm worried about the impact in users who enable that feature not knowing what the default Klipper behavior with it (progress stuck to 0) will be.

supermerill, 2022-03-07:
> However, I'm worried about the impact in users who enable that feature not knowing what the default Klipper behavior with it

I can move the "silent mode" from advanced to export mode.
Klipper profiles in the wizard don't have it activated, to begin with.

pedrolamas, 2022-03-08:
Well, PrusaSlicer replied with "Then disable the stealth mode in PrusaSlicer." and closed the issue...

In the meantime, after further discussions in Klipper Discord, I've open a PR in Klipper repo to get this sorted directly there https://github.com/Klipper3d/klipper/pull/5333

pedrolamas, 2022-03-08:
Good news: Klipper PR has been accepted and merged! 🙂

@supermerill Will leave this open and let you decide if you want to do anything or close as is!

closed 2022-03-11 by supermerill