supermerill/SuperSlicer #2484: Slicing shells/perimiters bugged

Exerqtor, 2022-03-07:
### What happened?

Trying out 2.4.58.1 and i'm getting some weirdness with the slicing.
Everything looks good in the "Sliced preview" tab but when i "Gcode preview" some perimiters are randomly missing.

[image]

### Project file & How to reproduce

[Shape-Box-slicing_bug.zip](https://github.com/supermerill/SuperSlicer/files/8198820/Shape-Box-slicing_bug.zip)

### Version

2.4.58.1

### Operating system

Win10 x64

Exerqtor, 2022-03-07:
Sliced with seam position random:
[image]

Sliced with seam position rear:
[image]

supermerill, 2022-03-07:
you have checked `start_gcode_manual`
and you didn't add the relativev extrusion gcode.
add `{if use_relative_e_distances}M83 ; use relative distances for extrusion{endif}`

Exerqtor, 2022-03-07:
I have all that within in my print_start.g macro, and it haven't posed
issues in the previous versions 🤔

supermerill, 2022-03-07:
Oh right, i added something to add these lines for the gcode preview. I'll look why it's not active anymore

nicedevil007, 2022-03-13:
This issue is gone with the latest nightly in my case, ty @supermerill

closed 2022-07-14 by supermerill