supermerill/SuperSlicer #2488: Can't open Superslicer 2.4.x on Mac OS because Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib

MarcoZ76, 2022-03-07:
### What happened?

Can't open Superslicer 2.4.x on Mac Os Catalina (not ARM)

Process:               SuperSlicer [55687]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.1_2022-03-07 (SuperSlicer_2.4.58.1_2022-03-07)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [55687]
User ID:               501

Date/Time:             2022-03-07 21:20:37.722 +0100
OS Version:            Mac OS X 10.15.7 (19H1715)
Report Version:        12
Anonymous UUID:        434BC4A2-71CA-0176-1299-D31213FCE15B

Sleep/Wake UUID:       F3F28DCF-AB13-4A90-8861-055BA634FA73

Time Awake Since Boot: 140000 seconds
Time Since Wake:       1500 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
  Referenced from: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
  Reason: image not found

Binary Images:
       0x10533c000 -        0x10717bfc7 +com.merill.slic3r/ (SuperSlicer SuperSlicer_2.4.58.1_2022-03-07 - SuperSlicer_2.4.58.1_2022-03-07) <31260B14-E552-31BC-8629-7B69F306FE1A> /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
       0x108a40000 -        0x108ad2267  dyld (750.7) <9FBB038E-684F-33E1-95A4-6B2CCC830B1B> /usr/lib/dyld
    0x7fff2f8ed000 -     0x7fff2fd6eff6  com.apple.CoreFoundation (6.9 - 1678.103) <B00C545B-7AAD-3E7B-8B80-357951CB6A21> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
    0x7fff3270a000 -     0x7fff327aeff3  com.apple.framework.IOKit (2.0.2 - 1726.148.1) <0A9D244C-AECC-3D29-86DE-4F3B04F8DD25> /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit
    0x7fff3a2c8000 -     0x7fff3a2d7ff7  com.apple.opengl (17.10.22 - 17.10.22) <251345CF-9DE1-342A-8710-2543AEA59B18> /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL
    0x7fff665e2000 -     0x7fff665e3fff  libSystem.B.dylib (1281.100.1) <9CF4DBEB-7D02-3C6D-8239-9DF979C24861> /usr/lib/libSystem.B.dylib
    0x7fff66951000 -     0x7fff6695dfff  libbz2.1.0.dylib (44) <14CC4988-B6D4-3879-AFC2-9A0DDC6388DE> /usr/lib/libbz2.1.0.dylib
    0x7fff6695e000 -     0x7fff669b0fff  libc++.1.dylib (902.1) <59A8239F-C28A-3B59-B8FA-11340DC85EDC> /usr/lib/libc++.1.dylib
    0x7fff67454000 -     0x7fff67544fff  libiconv.2.dylib (59) <18311A67-E4EF-3CC7-95B3-C0EDEE3A282F> /usr/lib/libiconv.2.dylib
    0x7fff677b8000 -     0x7fff677d0ff3  liblzma.5.dylib (16) <C131EF18-2CDD-3271-8A30-A8760D4FE166> /usr/lib/liblzma.5.dylib
    0x7fff68bae000 -     0x7fff68bc0ff3  libz.1.dylib (76) <793D9643-CD83-3AAC-8B96-88D548FAB620> /usr/lib/libz.1.dylib

Model: iMac13,2, BootROM 425.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 3,4 GHz, 32 GB, SMC 2.11f14
Graphics: kHW_NVidiaGeForceGTX680MXItem, NVIDIA GeForce GTX 680MX, spdisplays_pcie_device, 2 GB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x859B, 0x435438473353313630424D2E4D3136464544
Memory Module: BANK 0/DIMM1, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434314753364D465238432D50422020
Memory Module: BANK 1/DIMM1, 8 GB, DDR3, 1600 MHz, 0x859B, 0x435438473353313630424D2E4D3136464544
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xF4), Broadcom BCM43xx 1.0 (7.21.190.33 AirPortDriverBrcm4360-1601.1)
Bluetooth: Version 7.0.6f8, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: APPLE HDD ST1000DM003, 1 TB
Serial ATA Device: APPLE SSD SM128E, 121,33 GB
USB Device: USB 3.0 Bus
USB Device: Elements 25A2
USB Device: USB3.0 Hub
USB Device: Elements 10B8
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB2.0 Hub
USB Device: 3Dconnexion Universal Receiver
USB Device: QL-570
USB Device: SMI
Thunderbolt Bus: iMac, Apple Inc., 23.4

### Project file & How to reproduce

[Report.txt](https://github.com/supermerill/SuperSlicer/files/8200623/Report.txt)

### Version

2.4.58.1

### Operating system

Mac OS  10.15.7

### Printer model

Prusa

supermerill, 2022-03-07:
That new dependencies is a bit cumbersome. It shouldn't be needed...
For the meantime, you may have to install zstd with homebrew.
[execute this command](https://formulae.brew.sh/formula/zstd): `brew install zstd`

ocochodzi111, 2022-03-09:
> 

same here. (mac version on mac arm)

liftbag, 2022-03-09:
Duplicate of issue #2443

CrazyCreator13, 2022-03-09:
same problem here

hagan66, 2022-03-13:
Same problem Mac OS 11.6.4 intel.

liftbag, 2022-03-16:
> Same problem Mac OS 11.6.4 intel.
> 
> HOWEVER, after installing brew (failing once, asked to install xcode command line tools, which I did, and then Brew again) it works!

Can you run SuperSlicer 2.3 after editing and saving any profile from SuperSlicer 2.4?

hagan66, 2022-03-19:
Apparently not, this happened

bakaufman, 2022-04-06:
> That new dependencies is a bit cumbersome. It shouldn't be needed... For the meantime, you may have to install zstd with homebrew. [execute this command](https://formulae.brew.sh/formula/zstd): `brew install zstd`

worked fine for me - 12.3.1 IMac Retina 4k 3.6 GHz Quad-core Intel Core i3

adawalli, 2022-04-12:
I installed the command via brew as requested - however, zstd doesn't get soft-linked into /usr/local/opt/zstd/, so this fix doesn't really help as linked

adawalli, 2022-04-12:
If you are really anxious to make this work...

```bash
brew install zstd
sudo ln -s /opt/homebrew/Cellar/zstd/1.5.2/lib/libzstd.1.5.2.dylib /usr/local/lib/libzstd.1.dylib
```

If you are running on an m1 mac, the wrong bottle will have been fetched above...so
```bash
brew uninstall zstd
brew fetch --force --bottle-tag=x86_64_monterey zstd
```
Pay attention to where the bottle was downloaded...and then run something similar to the following
```bash
brew install /Users/<YOUR_USERN>/Library/Caches/Homebrew/downloads/5ccc6e72496939f946fe968cd3ddfc1ae09b8c032c9f49fa83e4ff31b33c3342--zstd--1.5.2.monterey.bottle.1.tar.gz
```
You can then open the latest slicer

CrazyCreator13, 2022-04-13:
will there soon be a solution without the extra installation of brew?

supermerill, 2022-04-13:
> will there soon be a solution without the extra installation of brew?

I don't know what to change to avoid that. I think I have to force to link with the .a instead of the .dylib ... but as I don't have a physical access to a not-arm mac, it's too time consuming for me to tinker.

If someone find a way to compile it statically on the github runner, please tell me how.

adawalli, 2022-04-13:
> I don't know what to change to avoid that. I think I have to force to link with the .a instead of the .dylib ... but as I don't have a physical access to a not-arm mac, it's too time consuming for [...]

@supermerill - one issue that I still see is that you are linked to dylib at a location that isn't even used by the latest brew install (at least on my machine) - I don't see the brew install as that big of a deal, but we ought to figure out how to get the path fixed correctly - let me know if you need help testing

adawalli, 2022-04-13:
ahh, so I see - on M1 macs...the base directory has changed to "/opt/homebrew/lib/ "...see https://docs.brew.sh/Installation

supermerill, 2022-04-13:
if you're using arm, here is a 2.4 build
https://mega.nz/file/9gwyHYKS#-BJPvMGzupZPLbnMqWBF3elMV7BLzXXqnBZ6IXNO2HU

CODeRUS, 2022-04-14:
> if you're using arm, here is a 2.4 build https://mega.nz/file/9gwyHYKS#-BJPvMGzupZPLbnMqWBF3elMV7BLzXXqnBZ6IXNO2HU shouldn't need zstd

I have Macbook air m1 with macos 12.3.1
attached build cannot be launched, os says its damaged.

supermerill, 2022-04-14:
ha, yes, i have to sign it.
You should be able to run it if, after dropping the app in your `Applications`, by doing `xattr -cr /Applications/SuperSlicer.app` in a terminal before executing it.

CODeRUS, 2022-04-14:
@supermerill works after xattr trick, thanks!

CrazyCreator13, 2022-04-14:
and for normal Mac with intel cpu?

supermerill, 2022-04-14:
> and for normal Mac with intel cpu?

until someone found why it's not compiled statically, you have to `brew install zstd`

CrazyCreator13, 2022-04-14:
that should also work without brew. it worked all the time... it works with the prusa slicer too. 
installing any extra things is more than impractical

supermerill, 2022-04-14:
> it worked all the time...

prusa added zstd as dependency with 2.4
I don't know how they build their 2.4, but that's the only practical way I found.

CrazyCreator13, 2022-04-14:
Well ...  then I'll wait

JWRu, 2022-06-22:
I have a iMac (Intel) Retina 5K, 27-inch, 2017 running macOS Monterey 12.4
Installing zstd via brew does not prevent Superslicer 2.4 from crashing at startup.

supermerill, 2022-07-15:
I tooke the time to setup the signing process, so you shouldn't need it anymore.
But it seems like the arm one I uploaded wasn't signed (I used the wrong one?). Now it should be good... hopefully.

Tenttu78, 2022-07-17:
I have this same named somekind beta version what works now, but now it doesnt open this new version.. What i need to do?

Process:               SuperSlicer [8507]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.3_2022-07-13 (SuperSlicer_2.4.58.3_2022-07-13)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-07-17 18:52:09.8187 +0300
OS Version:            macOS 12.4 (21F79)
Report Version:        12
Anonymous UUID:        2396307F-890D-6C1E-ADD3-E2E04CE52B69

Time Awake Since Boot: 38000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file)
(terminated at launch; ignore backtrace)

KenK73, 2022-10-23:
After installing brew and zstd, I am running SuperSlicer v2.4.58.5 successfully on my intel Mac mini.

masterleo, 2022-12-10:
I can confirm too that on my mac from 2019 with mac os 12.6 montery, the problem has arise and is simply corrected by closing the error report and doing :
brew install zstd
then reopening superslicer.app again. 
I can give a remote acces to my mac if a developer from the team wants to try out so stuff on my mac. 
With team viewer :) don't esitate to tell me at leoetchaas (ate) gmail dot the com

d4mer, 2022-12-14:
I can provide my Mac as well. I really need this to work without work arounds. Any thing I can do to help, please let me know

d4mer, 2022-12-14:
I just found this if it helps:

LibArchive pointing to a nonexistent path

If a packaged Slicer is launched on another mac and it crashes with the error log saying that

```
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
  Referenced from: Slicer.app/Contents/lib/Slicer-4.13/libarchive.17.dylib

```
It means that libarchive has has found homebrew versions of some of it’s requirements, rather than local ones. For the packaged version of Slicer to run on other machines none of the prerequisites should be installed via homebrew. For example lz4 and zstd are bundled with subversion and rsync so if you have these two application installed via homebrew, libarchive will grab them from /usr/local/opt/ and the packaged Slicer will not run on other machines. The solution is either to remove them from from homebrew with brew remove lz4 and brew remove zsdt or to change the $PATH so that the local build folder goes before /usr/local/opt/. After doing this Slicer should be rebuilt and repackaged. 

See [Relevant issue that’s tracking this error](https://github.com/Slicer/Slicer/issues/5415)

Lupo-Grigio, 2023-01-15:
Jan 14 can (re-Confirm)
2019 intel macbook pro 2019 
Big Sur 11.6
problem fixed by opening terminal window and typing
brew install zstd

d4mer, 2023-01-15:
> Jan 14 can (re-Confirm)
> 
> 2019 intel macbook pro 2019 
> 
> Big Sur 11.6
> 
> problem fixed by opening terminal window and typing
> 
> brew install zstd
> 
> 

That is definitely not a fix. It is a workaround at bast

CrazyCreator13, 2023-01-15:
i think too ... all other slicers, don't need this workaround.

d4mer, 2023-03-05:
Hi All,

As no one wanted to compile 2.4 for macOS correctly, I have done so myself. I am not a programmer and don't really know what I am doing but if someone would like to test it out, please drop me a message and I will happily share the dmg with you. Also, If someone could point out how to make it available on GitHub, let me know so I can upload it

Regards,

Ed

masterleo, 2023-03-07:
please send to leoetchaas@gmel.com (typo)