supermerill/SuperSlicer #2489: Cannot switch to darkmode

foreachthing, 2022-03-08:
### What happened?

Switching to darkmode causes crash:

```
 ---------------------------
SuperSlicer
---------------------------
Unhandled unknown exception; terminating the application.

---------------------------
OK   
---------------------------

```

[image]

### Project file & How to reproduce

1. Start SuperSlicer
2. Go to Preferences -> "Colors and Dark mode"
3. Enable Dark mode -> crash.

### Version

2.4.58.1

### Operating system

Win106x4

### Printer model

UM2

foreachthing, 2022-03-08:
Dup! Sorry!

closed 2022-03-08 by foreachthing

supermerill, 2022-03-08:
should be fixed in nightly