supermerill/SuperSlicer #2490: ui_layout is overwritten at startup

foreachthing, 2022-03-08:
### What happened?

* `ui_layout = ANYTHING_HERE` gets overwritten with `ui_layout = Standard` after every restart. Except, when changing the settings to `ANYTHING_HERE` and requested restart. Then it works. Restarting application "manually" resets settings.

### Project file & How to reproduce

1. Change UI-Layout to non-Standard.
2. Let it restart.
3. Restart manually and check settings.

### Version

2.4.58.1

### Operating system

Win 10 x64

### Printer model

UM2

supermerill, 2022-03-08:
can't reproduce.
can you send me your conf folder?

foreachthing, 2022-03-08:
As raw as it gets: [SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/8206287/SuperSlicer.zip)

I still can reproduce, even after a complete deletion of the profile folder.
PS: After deleting the SuperSlicer profile folder, SuperSlicer wouldn't start anymore.
[image]

supermerill, 2022-03-08:
launched a nightly build with the fixes.

foreachthing, 2022-03-08:
Awesome!! 👍🏼 
Thanks a bunch!

closed 2022-03-08 by foreachthing