supermerill/SuperSlicer #2491: Expert mode lacks some features present in Advanced mode

macchinette, 2022-03-08:
### What happened?

Hi Merill, I have seen that the settings for the seam position are different between the modes Simple / Advanced / Expert.

For the seam position, I would have expected that in Simple mode there was no custom function as it is missing the settable fields.

For the seam position, I would like all the functions of the Advanced mode to be in Expert mode and handled in the same way.

I hope it is possible to do.
Below photos to understand.

Thank you

### Project file & How to reproduce

Simple mode
[image]

Advanced mode
[image]

Expert mode
[image]

### Version

SuperSlicer_2.4.58.1_2022-03-07

### Operating system

Windows 10 Pro 21H2

### Printer model

Anycubic I3 Mega S

supermerill, 2022-03-08:
Corners and nearest are just pre-defined cost. 
"Personalized" is a way to say "it's not a pre-defined cost, something else".

In expert, as you're an expert, you can enter the cost yourself and you don't need these pre-defined helpers, that's the thought behind it.

the tooltip is a bit succinct...

It's not set in stone, more like an experiment. Tell me what I should do to avoid confusion.

macchinette, 2022-03-08:
> Corners and nearest are just pre-defined cost. "Personalized" is a way to say "it's not a pre-defined cost, something else".
>
> In expert, as you're an expert, you can enter the cost yourself and you [...]

You are right that it is in Expert mode, but to explain the function in detail it would be complicated to do it with a tooltip to be translated into various languages, I would make the setting as in Advanced mode, at least with the two default settings corners and nearest you could have an example of how to use the custom setting.
In addition I ask you how the two cost fields work, must the total always have to be 100?
Because I tried to use it, but didn't understand the function much.
Thank you

supermerill, 2022-03-08:
In addition I ask you how the two cost fields work, must the total always have to be 100?

if you put 0 to cornern it won't try to hide the seam in a corner.
if you put travel to 0 it will disregard the travel time from the last point (often in infill for the external perimeter) and the seam.

if they aren't 0, it will choose a compromize: a corner not too far away.

There is no upper limit. Some other weight exists and can be trivialized if you put very hight cost here. It's made to have something in the 50-100 range.

macchinette, 2022-03-09:
> In addition I ask you how the two cost fields work, must the total always have to be 100?
>
> if you put 0 to cornern it won't try to hide the seam in a corner. if you put travel to 0 it will disregard [...]

Thank you for the explanation, I will try to make good use of it, however I think it is more intuitive to have the same management of Advanced if it does not bother you too much.

Thank you

supermerill, 2022-03-09:
np, thanks for the feedback

closed 2022-07-14 by supermerill