supermerill/SuperSlicer #2496: Infill in the place of Perimeter

dingyifei, 2022-03-08:
### What happened?

Super slicer infill behaves differently compared to Prusa slicer, where SS has this layer of infill replacing the entire perimeter when top/bottom is removed. The following is in SS:
[image]
The following is in Prusa slicer with the same 3MF file (minor adjustments on printer settings):
[image]

### Project file & How to reproduce

[13Pro Case.zip](https://github.com/supermerill/SuperSlicer/files/8209230/13Pro.Case.zip)

Slice without top/bottom and several loops of infill

### Operating system

Fedora 35

### Printer model

Prusa MK3S+

neophyl, 2022-03-09:
You have Perimeters & Shell>Quality>One One Perimeter>On Top surfaces enabled.  Turn it off and it slices similar to PS.
That option and setting doesn't exist in PS and its turned on by default in SuSi.  

Whenever there is an issue like this there are always certain settings I check, single perimeter is one of them as it often seems to cause weird edge cases.  About as often as Detect thin walls does in PS.

dingyifei, 2022-03-09:
> Whenever there is an issue like this there are always certain settings I check, single perimeter is one of them as it often seems to cause weird edge cases. About as often as Detect thin walls does in PS.

Thank you! it seems to be the cause. I guess it's just a side effect of single perimeter lol

closed 2022-03-09 by dingyifei