supermerill/SuperSlicer #2497: Option to show/remove Grid from the objects

kanawati975, 2022-03-08:
I'm using the Release 2.4.58.1 on Windows 10 and it's faster and less crashes when slicing multiple objects.

However, there is a back Grid showing on the objects, and it's kind of annoying if one would like to examine Layers.
Even when using a texture on the build plarform does not hide this Grid.

I hope that there is an option to show/hide this grid

[image]

[image]

[image]

tome9111991, 2022-03-08:
Hi.
Try the latest nigjtly
Should fix this bug

kanawati975, 2022-03-09:
Great!
Thank you <3

closed 2022-03-09 by kanawati975