supermerill/SuperSlicer #2498: UI Graphical issues

superblobmonster, 2022-03-09:
### Version
Screenshots are all from 2.3.57.11, also happens in 2.4.58.1

### Operating system type + version

[image]
[image]
* GUI Settings:
[image]

### Behavior
A majority of the non-resizable windows have cropped content and inaccessible options.
Some of the normal slicer options have text boxes that are too small, strange text wrapping, and partially cropped buttons.
[image]
[image]
[image]
[image]

supermerill, 2022-03-09:
maybe related: #1054 #1698

supermerill, 2022-03-14:
gtk2: checkboxes are too small
gtk3: +- on int types are messing everything as they are huge.

I'll remove the +- on gtk3, to avoid problems (I don't have the weeks needed to ensure all the GUI is handling these widget perfectly)
I'll see if I can do something for the checkboxes on gtk2

neophyl, 2022-03-14:
I recall that PS had all sorts of issues around this area too and they are now only supporting their specific app images, so an issue needs to be applicable to those rather than anything self compiled.
Not sure if they are using specific versions of gtk2/3 that have any modifications.  

Not really up on Linux but do tend to monitor the PS github :)

supermerill, 2022-03-14:
Found a fix for gtk2, using toggle buttons instead of checkbox (when the zoom is too high).

supermerill, 2022-03-14:
for gtk3, I just increased a bit the size of the widget.
Will be in next nightly, please report.

superblobmonster, 2022-03-15:
I don't see a nightly for Linux yet, will test when appimage is available

supermerill, 2022-03-16:
sorry, one fix doesn't compile on linux. And it's much more tricky to make it compile as intended.

superblobmonster, 2022-03-25:
Fixed in Nigthly ubuntu #522

closed 2022-07-14 by supermerill