supermerill/SuperSlicer #2499: Wall Overlap Rate

XSJFYZS, 2022-03-09:
Hi，Thank you very much for your dedication
I can't change the overlap ratio of the inner wall to the outer wall
The max can set to 100%
Perfect printing with PLA, but the TPU separates
not glued together，changing the width didn't help either
Maybe I'm missing something, any help thanks
[image]
[image]

supermerill, 2022-03-09:
That "overlap" is here to be able to reduce over-extrusion for filaments that are not fuild enough or when you want a little under-extrusion for the perimeters.
You can't go higher than 100% because that would create over-extrusion.

If you have flow problem, please don't touch them. Use the filament extrusion multiplier.

XSJFYZS, 2022-03-10:
But I have no problem using PLA, also calibrated extruder and flow
any other solution?

> That "overlap" is here to be able to reduce over-extrusion for filaments that are not fuild enough or when you want a little under-extrusion for the perimeters. You can't go higher than 100% because [...]

neophyl, 2022-03-10:
Flexible filaments dont generally feed the same which is why often the extrusion multipliers for those filaments are different.  Just because its nice with PLA doesnt mean a different filament will be.