supermerill/SuperSlicer #2500: First click in the 3D display doesn't behave as intended

sarchar, 2022-03-09:
### What happened?

The first click in the 3D area of the "3D view" does not handle the mouse click properly.  Instead of dragging the object around the build plate, as expected, the view is rotated. 

### Project file & How to reproduce

0. Make sure the 3D View tab is active

1. Go to file->Import->Import STL/OBJ/AMF/3MF...

2. In the file explorer dialog, *double click* on an STL file to import it. It is important to double click a file, not use the OK button.

3. Try to drag the object around the build plate. The view rotates instead.

Also, try rotating the world, the view will snap quickly before rotating smothing.

### Version

2.3.57.7

### Operating system

Windows 11

### Printer model

n/a

m0n5t3r, 2024-03-26:
Version 2.5.59 Appimage does the same thing - clicks on the object(s) don't get recognized, so one can't drag the part around, place on face, rotate, etc.; buttons on the left work if the object is selected, but once you try to interact with it it gets deselected.

m0n5t3r, 2024-03-26:
Tested with the latest nightly as well, same thing

SuperSlicer 
Version:   2.5.59.9
Build:     SuperSlicer_2.5.59.9_2024-03-22

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 6.5.0-26-generic x86_64
Total RAM size [MB]: 32,843MB
OpenGL installation
GL version:   4.6 (Compatibility Profile) Mesa 24.1~git2403230600.b83578~oibaf~m (git-b835784 2024-03-23 mantic-o
Vendor:       AMD
Renderer:     AMD Radeon Graphics (radeonsi, rembrandt, LLVM 16.0.6, DRM 3.54, 6.5.0-26-generic)
GLSL version: 4.60
<details>
<summary>Installed extensions:</summary>
GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_multisample_advanced
GL_AMD_multi_draw_indirect
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
[... 241 lines cut ...]
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_NV_vdpau_interop
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
</details>