supermerill/SuperSlicer #2503: question about retraction point

CuredPrusa, 2022-03-09:
### Version
Version: 2.3.57.11
Build: SuperSlicer_2.3.57.11
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

### Behavior
Is it somehow possible to adjust the retraction point after perimeter is done?
in this example, one wall is set to three perimeters. The model is straight pipe. Wall of the model is 6 perimeters thick, so it'll have six lines. Printing is set inside out. Inner wall ends and retracts somwhere on the inner perimeter. The seam on the top of model looks bulgy. I think that changing this retraction point to the actual end of the line would make smoother seam in the top plane.  Am I missing some settings which can regullate the retraction point? I have also noticed that even without retraction the nozzle travels over previous perimeter a bit. I understand it fills the gaps of a seam this way, to make seam stronger. But can that be adjusted somehow?

[image]

CuredPrusa, 2022-03-09:
I see that this is called wipe. 
I can only increase that additional travel, because wipe is actually turned off.

CuredPrusa, 2022-03-11:
Please add it to modifiers

supermerill, 2022-03-11:
If I add it to the modifers, i have to add also all filaments & extruder settings.

CuredPrusa, 2022-03-11:
It would be nice to have all those settings in modifiers.
I don't think that Retraction is Printer /Extruder setting. It should be adjusted according to filament type, speed, temperature. 
"Cooling" is also related more to the speed than Filament.
There are many many advatages of Slic3r successors over Cura, but in this case, the points go to Cura because all settings are in one cluster, in one profile. I think most of parameters are too much related, to just cut them off to another standalone parameter group, that's not convenient. I don't ask you to add all this to modifiers if it's too much work. I understand. Some workarounds can be used for compensation of incomplete modifiers choice. 

But if you find inspiration to do it, that would be mostly welcome.