supermerill/SuperSlicer #2504: Strange behavior when resizing a modifier

robustini, 2022-03-09:
### What happened?

As you can see in the video I link when resizing a modifier it doesn't stay in the same position as it should.

### Project file & How to reproduce

[FUSO05.zip](https://github.com/supermerill/SuperSlicer/files/8215773/FUSO05.zip)

https://user-images.githubusercontent.com/3864712/157470695-c7de250f-1b97-45fe-a25e-62795825bebc.mp4

### Version

2.4.58.1 nightly of today

### Operating system

Windows 10 Pro x64

### Printer model

SP

robustini, 2022-03-10:
There is also another very boring thing, when you resize the cylinder by taking it from the orange cubes, the ones on the corners, you not only resize the diameter but also the height.
It would be useful if it dimensioned only the diameter, leaving the height unchanged.

https://user-images.githubusercontent.com/3864712/157615748-58f8b0e5-87af-414c-92d8-37ad35d90d68.mp4

supermerill, 2022-03-10:
Also, when the main object is resized, the parts/modifier resize text boxes won't work properly, as it's scaled by the root object.

neophyl, 2022-03-10:
Please don't change the orange resize function.  It does exactly what its supposed to be doing.  The orange handles resize the object globally, the others just resize a specific axis.

supermerill, 2022-03-10:
> Please don't change the orange resize function. 

I think that the problem is more the translation than the scaling with the 3d handles.

robustini, 2022-04-04:
Fixed in the nightly, thanks!

closed 2022-04-04 by robustini

supermerill, 2022-04-04:
I didin't do anything. Maybe resolved by the prusa 2.4.1.

supermerill, 2022-04-04:
oh, no that I think of it, it's probably because of 4c3749b1