supermerill/SuperSlicer #2509: Per layer configuration change

KERFMAS4, 2022-03-10:
The simulation bar at the right where you can change the color right now. It would be amazing if there is a button like the color change, but to make a configuration change. For instance, you print in normal mode until specified layer, lets say two legs, then you add the vase mode for the rest of the print. Could be nice it it lets you change almost any conguration at any layer height. I should have to make the simulation when you add any parameter, so could be great to have a list box in the right panel with the layers and the config change from that layer.
Thank You!

KERFMAS4, 2022-03-10:
Its ok. But I think its faster and easier in the sliced preview tab. Right clicking the + button. Also seems vase mode can't be modified right now.

supermerill, 2022-03-10:
I may be able to set a shortcut for layer range creation on the (+) menu, should it be enough?

KERFMAS4, 2022-03-10:
Yes please, Thanks a lot.

trysten, 2023-05-27:
I have found myself obsessing with bridging layers and targeting many settings to a single layer. I would like to see this feature. I also wanted to spam some gratitude to merill: thank you so much for all your contribution. I don't have any luck with these kinds of bridges on other slicers. 
[image]