supermerill/SuperSlicer #2510: Error encountered "Invalid value provided for parameter Thin_perimeters: 0%"

Sirnut, 2022-03-10:
### What happened?

I installed the beta recently, but otherwise did nothing besides open superslicer up to this error:

[image]

After installing the beta, I did make the configuration change to enable dark mode, but otherwise nothing should have changed. I install official releases with chocolatey, and installed the beta release manually in a different directory

### Project file & How to reproduce

[0.20 PETG NORMAL.zip](https://github.com/supermerill/SuperSlicer/files/8225473/0.20.PETG.NORMAL.zip)

This is one of my configurations

### Version

2.3.57

### Operating system

Win10

supermerill, 2022-03-10:
that's when you open a 2.4.58 project.
Will do something for the next 2.3 to have some sort of failback.

Sirnut, 2022-03-11:
Please do because the error happens when I try to open a brand new 2.3 instance so I am unable to open the slicer currently...
I don't even know how to resolve this issue honestly. Using the launcher from my chocolatey folder is what produces this error. The install directory for my 2.3 and 2.4 programs were supposed to be entirely different though. Why is this happening?

supermerill, 2022-03-11:
You can create a folder named "configuration" next to the exe.
You can append --datadir X:\your\profile\location to the target properties of a shortcut to the executable

Sirnut, 2022-03-11:
Similar to this? It's just an empty folder but it seems to still be throwing the errors
[image]

This is the shortcut I use to launch SS, how do I modify the properties to try and load the blank configuration file?

supermerill, 2022-03-11:
a bit too complicated with chocolatey, you have to create a not-admin dir in lib/[...]/superslicer
the --datadir option is simpler
[image]

Sirnut, 2022-03-12:
Thank you for the walkthrough!! I was able to load 2.3 :) Albeit without some of my more recent settings, but those wont be too hard to copy over from the 2.4

You're the best Merill