supermerill/SuperSlicer #2512: wxWidgets Debyg Alert timerimpl.cpp(60) timer can only be started from the main thread [in tread 330]

FlatEarthDave, 2022-03-11:
### What happened?

This error:
[image]

### Project file & How to reproduce

Calibration >> bridge flow calibration >> test flow >> G>

### Version

2.4.58.1

### Operating system

windows 11

### Printer model

custom corexy with klipper

supermerill, 2022-03-11:
Can't reproduce.
Is it easy to reproduce on your side or it's really random?
Do you use the "background processing"?
I may have to send you a special version to have a better log of the error.

FlatEarthDave, 2022-03-11:
Yes it is extremely easily reproduced. I had the same issue with the initial 2.4 beta release, so I went back to 2.3.57 and it went away. It doesn't seem to affect the print, but it does cause a random crash.

Also, it looks like 2.4 doesn't respect the "complete each object", or at least it didn't for the bridging tests, even though the option was checked.

I would be happy to test a different version to help with debugging it.

julianschill, 2022-03-19:
I experience the same when uploading to Moonraker: 
[image]
 2.4.58.1 Nightly
 Background Processing switched off.
 
[image]

FlatEarthDave, 2022-03-20:
It actually went away after a few times and it hasn't reappeared. I didn't change any settings or builds.

[image]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F13122110%2F159122131-395a703a-8fad-4eb6-9b0a-eccd0531059a.png&data=04%7C01%7C%7C963b1e5b90814d7ab70b08da09a8f708%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637832918546034260%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=kOUq3fZc9KJbkdg8%2BBcna6MigpaJxaOC30GmpdWN2IE%3D&reserved=0>

[image]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F13122110%2F159122188-5ee3e60e-9329-48fd-9e0f-5583232bdff9.png&data=04%7C01%7C%7C963b1e5b90814d7ab70b08da09a8f708%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637832918546034260%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=c%2FyE5n1cfA9WFBlH6pyljp6qxuprrYo2%2FFCwzYlAF4M%3D&reserved=0>

supermerill, 2022-03-26:
can someone use this debug build? the error message should be better.
https://github.com/supermerill/SuperSlicer/suites/5802201336/artifacts/193999699

julianschill, 2022-03-27:
Tried with the debug build and got this message: 
[image]

supermerill, 2022-03-31:
can someone test this nightly? https://github.com/supermerill/SuperSlicer/suites/5835274444/artifacts/196041452
it should resolve the issue.

closed 2022-07-14 by supermerill