supermerill/SuperSlicer #2513: slice even if sliced

CuredPrusa, 2022-03-11:
### Version
Version: 2.3.57.11
Build: SuperSlicer_2.3.57.11
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

### Behavior
Sometimes after slicing my buildplate turns blank (only in Gcode preview).
Then I have to reslice it to bring it back. But to do that, i have to change some parameter.
It would be nice if Ctrl+R would slice even if "sliced" flag is present.

supermerill, 2022-03-11:
duplicate of #1532 (and #2385)
fixed in next nightly (maybe this night)

CuredPrusa, 2022-03-11:
Please show me where can I find some instructions how to dowload and install the "nightly" version.

supermerill, 2022-03-11:
Click on the "Actions" menu
if you're on windows, click on the first title above a "C/C++ Nigthly windows x64"
than click on nightly_win64