supermerill/SuperSlicer #2514: Custom gcode for wipe tower

mann1861, 2022-03-11:
**Is your feature request related to a problem? Please describe.**
Certain settings such as pressure advance get overwritten when doing wipe tower. It would be useful to be able to add custom gcode before and after wipe tower. Tip cooling moves can be confused as clogs by filament encoders.

**Describe the solution you'd like**
On printer settings custom gcode page a box to enter custom gcode to be inserted before and after wipe tower gcode. Perhaps only enable for machines with multiple tools

**Describe how it would work**
Current wipe tower looks like this and custom gcode should be inserter as first command
`;HEIGHT:0.2`
`;TYPE:Wipe tower`
`;WIDTH:0.5`
`;--------------------`
`; CP TOOLCHANGE START`
`; toolchange #325`
`; material : ABS -> ABS`
`;--------------------`
`CUSTOM GCODE HERE`
`M220 S100`
`; CP TOOLCHANGE UNLOAD`
`;WIDTH:0.6`
`M73 P2 R783`
`G1  X236.108 Y219.124  `
`SET_PRESSURE_ADVANCE ADVANCE=0`

Seperate custom gcode should also be applied after end of wipe tower

**Describe alternatives you've considered**
Pressure advance can be worked around by applying it at filament level but more complex macros are required when using different nozzle sizes.

HelgeKeck, 2022-05-22:
i second this

moo00se, 2022-07-12:
This "issue" has not been addressed in v2.4