supermerill/SuperSlicer #2515: Elephants foot compensation feature

ModMike, 2022-03-11:
Would it be possible to build an elephants foot compensation feature to reduce the to use rafts? There may be betters ways to do it than I am envisioning but this is what I was thinking:

Setting 1: Number of **Layers** to apply the compensation to

Setting 2: **Angle** to apply

Setting 3: Not necessary but could be useful, **Shape** of transition, curved or straight.

liftbag, 2022-03-11:
Elephant foot compensation is named XY first layer compensation in SuperSlicer. You have to put negative value to create an effective elephant foot compensation. Also, the Number of layers is already present, and the angle it's a consequence of the layer height, height in layers, and the compensation value.

[image]

ModMike, 2022-03-11:
Don’t I feel like a moron. Thanks for the lesson!

closed 2022-03-11 by ModMike