supermerill/SuperSlicer #2516: Nightly (Windows) - The M106 controls for the fan are completely wrong

robustini, 2022-03-11:
### What happened?

Inside the gcode I find many M106 S28.05, which would correspond to a percentage of 11%, never declared within the settings.
Furthermore, in the top layers (3 in my case) the fan is activated only in the last of the three, when it should do so in all the three top layers.
There are also M106 S2.55, that is the fan practically stopped.
It does not correspond to any setting declared by me as a percentage, considering that obviously fan at maximum would mean M106 S255.

[image]

### Project file & How to reproduce

Inside both the project and the exported gcode.

[FUSO05.ZIP](https://github.com/supermerill/SuperSlicer/files/8233550/FUSO05.ZIP)

### Version

2.4.58.1 Nightly

### Operating system

Windows 10 Pro

### Printer model

SP

tome9111991, 2022-03-11:
Full fan speed at layer 346 ?
Sounds s bit to much.

I use 5

robustini, 2022-03-11:
> 

If you do the slicing you will understand why.

supermerill, 2022-03-11:
1) as you set to gradually inscrese the fan speed from 0 at the first layer to 40% at the last layer, it's intended to have value between the two.
2) I'm not sure, but as the external perimeter fan speed is 0, if the cooling from short layer time triggered, it can increase it a bit, and so set it above 0.

robustini, 2022-03-12:
I misinterpreted the "full fan speed at layer", sorry.
But the problem of the top layers seems to me that there is, that is, it seems to me that the fan is activated only at the last of the top layers, not at the first.
It would also be useful to have the layer number in the gcode as an indicator of the height, it would be more intuitive.
Example: from ";Z:6.53" to ";layer:26", or a combo: ";Z:6.53 - layer:26".

robustini, 2022-03-19:
I did other tests and for me the problem is there.
With these settings the minimum value of M106 that I find in the gcode is "S196" from the fourth layer onwards.
But if I put "20%" in the outer perimeters, shouldn't it create "M106 S51.2"?

[image]

robustini, 2022-03-19:
Damn, I found where the problem lies, this is what messes it all up:

[image]

The layers in this print last less than 60 seconds on average, which is why I found the fan at most often.

closed 2022-03-19 by robustini