supermerill/SuperSlicer #2520: weird supports distance (not respecting value set)

aschor, 2022-03-12:
### What happened?

I have migrated all my settings from prusaslicer to superslicer, and discover that they are all "screwed" because the support distance does not behave the same with the same value ...

I open the same 3MF file in prusaslicer 2.3 and superslicer. support distance is set to 0.2mm (detachable).

YET. in prusaslicer, last support is printed at 31.30 mm height like here :
[image]

and this results in a distance of 0.6mm whith the layer at 31.9mm height (which is printed with bridge flow)

whereas in superslicer, with the SAME 3MF, the last support layer is printed at 31.5mm, so 0.2mm closer to the surface at 31.9mm... resulting in undetachable supports.

the same happens with the bottom distance of supports : 
15.50mm for the first prusaslicer support line here
[image]

15.30mm for the first superslicer support line here 
[image]

### Project file & How to reproduce

[bug.zip](https://github.com/supermerill/SuperSlicer/files/8237469/bug.zip)

### Version

2.3.57.10

### Operating system

windows 10

### Printer model

Hypercuve evolution

supermerill, 2022-03-12:
z-offset is "from plane" and not "from filament" as prusalicer uses.
I think it's a bad default from me. will change it.

supermerill, 2022-03-12:
Bad default on my side: I added the support_material_contact_distance_type setting, to let you choose the origin, adding a "from plane" option (prusaslicer uses 'from filament' or 'none/soluble' if the distance is zero).
I think the default should be 'from filament' to avoid confusion and problematic conversions.

closed 2022-07-14 by supermerill