supermerill/SuperSlicer #2521: Nightly (Windows) - The number of top layers is not always respected

robustini, 2022-03-12:
### What happened?

The number of top layers is not always respected.
In some conditions it removes a layer from those declared, for no apparent reason.

### Project file & How to reproduce

3 top layers declared and at that point it generates only 2.
At the top it generates 3 instead, as it should.

[image]

[FUSO05.zip](https://github.com/supermerill/SuperSlicer/files/8237994/FUSO05.zip)

### Version

2.4.58.1

### Operating system

Windows 10 Pro x64

### Printer model

SP

supermerill, 2022-03-12:
your modifier here modify the top layers to 2

robustini, 2022-03-13:
Damn sorry!

closed 2022-03-13 by robustini