supermerill/SuperSlicer #2524: Enhance the preset compare view with easy 'jump to setting' and granular copy/transfer settings button/options

netweaver1970, 2022-03-13:
**Is your feature request related to a problem? Please describe.**
no really related to a problem, just some extra handinesses I'm currently missing to make it even more super fantastic great ! :)
It would put preset maintenance to a whole different level. Now different presets will have setting differences, historically grown, not easy to rectify.
Also, since I just did a major toolhead+gantry upgrade, lots of tuning needs to be redone, I would like to keep presets now as much as possible in line with eachother.

**Describe the solution you'd like**
- default into a 'current open preset' on the left (fixed) and freely selectable preset on the right. With an checkbox to allow freely selectable presets on both side of the comparison, thus creating 2 major compare working modes. 
- when doing a preset compare (and you're in 'current open preset on the left' mode, it should be possbile to jump straight to the 'clicked-on' comparison result in your current open profile. That saves you from leaving the comparison manually, hunting down the setting in the fully preset hierarchy to -if needed- modify it.
- one step further, introduce a 'copy from comparison (aka right) preset' button, on every level of the comparison result hierarchy shown. THis should only be possible in the 'current open preset on the left' mode of course, as that one is really meant to 'work' on the left preset. With a confirmation window shown when selecting a non-single setting level (eg. Speed, or Speed-Modifiers) to prevent people of updating too many settings at once, if by accident they clicked a too high level.
It would be working like Midnight Commander, (Norton Commander for the older folks like me). Wth a limitation updates only can go from 'right to left' to protect the other existing preset from accidental modifications.

**Describe how it would work**
as above

**Describe alternatives you've considered**
All manual modifications. It works but now thanks to the comparison feature, one can see easyly where the diffences are. It's still rather labourous to make them all (the ones you want to be the same of course) equal, due to no 'jumping' to the setting or assisted copying.

**Additional context**
Add any other context or paint drawing about the feature request here.

supermerill, 2022-03-13:
what I propose:
* when you click on the name, it will send you to the setting, like for the search.
* when you click on a value (whatever the one), your current setting is switched to this one
* refresh without resetting the selected presets when you save your current preset.

netweaver1970, 2022-03-13:
Sounds OK to me. I guess this will be easier and quicker to implement than what I was proposing. Looking forward to it. Thanks!