supermerill/SuperSlicer #2525: inconsistent (with vertical gaps ) support generated

netweaver1970, 2022-03-13:
### What happened?

in the latest beta release, 2.58.1, I can see vertical gaps in generated support. Not always steady going up from the buildplate to the object.
I've indicated the gaps with yellow marker.
[image]

Even better visible on the 2nd image:
[image]

Supports are jumping around :) 

### Project file & How to reproduce

[support gaps.zip](https://github.com/supermerill/SuperSlicer/files/8239475/support.gaps.zip)

### Version

2.4.58.1

### Operating system

WIndows 10

### Printer model

Voron 2.4 with Mantis

supermerill, 2022-03-13:
Comes from the prusaslicer 2.4 changes.
Complicated to fix.
Can you ask in the prusaslicer github?

netweaver1970, 2022-11-04:
As per prusaslicers comment on my issue: works as designed 🤦

closed 2022-11-04 by netweaver1970