supermerill/SuperSlicer #2528: Failure when nozzle_diameter = 0.15

Hashan0, 2022-03-13:
### What happened?

Failure when nozzle_diameter < 0.195m
[image]
[image]
m

### Project file & How to reproduce

[config.zip](https://github.com/supermerill/SuperSlicer/files/8240275/config.zip)

### Version

2.4.58.1

### Operating system

windows 10

### Printer model

Custom

supermerill, 2022-03-13:
Can't reproduce with your config.
Can you post you **project file**?

Hashan0, 2022-03-13:
Creating Custom or others Printer
nozzle_diameter = 0.15 or 0.1
layer_hight = 0.05
first_layer_height = 50%
all other parameters are standard

Open Calibration cube
Change any parameter -> Failure:   Flow::spacing() produced negative spacing. Did you set some extrusion width too small?

config file:
[config.zip](https://github.com/supermerill/SuperSlicer/files/8240749/config.zip)
project file:
[Calibration cube.zip](https://github.com/supermerill/SuperSlicer/files/8240746/Calibration.cube.zip)

closed 2022-07-14 by supermerill