supermerill/SuperSlicer #2529: Sliced Preview Buggy

Alexander-T-Moss, 2022-03-13:
### What happened?

Once I slice a model it isn't showing many perimeter corners on parts, but then if I move the parts around and don't reslice the gaps are no longer their in the unsliced preview ( see photos for better explanation )

### Project file & How to reproduce

[image]
[image]

### Version

2.3.57.11

### Operating system

macOS Monterey 12.2.1

### Printer model

Voron 2.4

Alexander-T-Moss, 2022-03-13:
Update, I'm running a print now to see if this preview is representative of the actually gcode

supermerill, 2022-03-14:
the first screen is the slicing preview and the second one id the gcode preview.

Alexander-T-Moss, 2022-03-15:
So having printed off some parts, the missing gaps that can be seen in the gcode preview are not visible, so I'm assuming this is a bug and not me having messed up some settings :)

supermerill, 2022-03-16:
try to put a `G92 E0` in one layer change gcode.
in the 2.4, it's added automatically.

Alexander-T-Moss, 2022-04-02:
I have found the cause of the issue, I changed the settings below, was originally set to 2 on each
[image]