supermerill/SuperSlicer #2530: Brim ears should ignore support raft

Sirnut, 2022-03-13:
**Is your feature request related to a problem? Please describe.**
I have grown quite fond of brim ears to ensure adhesion for the duration of my prints, however an annoyance comes from the brim ears being attached to the raft automatically generated for supports
[image]

**Describe the solution you'd like**
I think Brim Ears should either ignore the raft generated by supports, or have a toggle. The rafts for supports already have great results with adhesion

**Describe how it would work**
When the slicer analyzes the 3D model for the addition of brim ears, I would just not include the solid layer produced by a raft in this analysis

Technofrikus, 2022-03-23:
Got the same problem. Want to prevent warping but the brim is not connected to the actual part because of the support.

PrusaSlicer does it correctly
[image]

SarraceniaWilds, 2023-04-09:
Ive found this bug to affect regular brim as well not just ears. Trying to produce a larger brim around the support material generates it on the outside. Inside brim does generate on the part and inside the support, but if you set the inside and outside to cover the surface entirely where the support is generated, it disappears. This is also true of the brim ears. Here are some screenshots as example
[image]
[image]
[image]