supermerill/SuperSlicer #2531: MPMDv2 profile

ricardodeazambuja, 2022-03-13:
I found a PR where it was added the possibility to connect directly to the MPMDv2. However, I couldn't find a profile for this printer. Am my missing something and looking for the profile in the wrong place? Thanks!

supermerill, 2022-03-14:
If nobody has created a vendor profile for the vendor of the MPMDv2, and didn't submit it, it can't be included.
If you search for the profiles/vendor files, they are at [slic3r/slic3r-profiles](https://github.com/slic3r/slic3r-profiles/blob/main/Creality.idx)

ricardodeazambuja, 2022-03-14:
Thanks for the reply @supermerill. I just thought the profile could be in the repo, hidden by a different name or something like that :)
@drbachler, since you were the author of #1807, would you mind sharing the profile?